Avnet/Ultra96-PYNQ #7: ultra96 sdsoc platform

jiangwx, 2018-09-20:
I had successfully built sdsoc hardware platform for zcu102 according to the guidence of ug1146 ,but I met some problem when building sdsoc hardware platform for ultra96.
I created vivado project from the vivado project in the ultra96.bsp and remove unrelated peripherals 
[image]
I created petalinux project from the ultra96.bsp too, and added xlnk in device tree, the system-user.dtsi is shown below
```sh
/include/ "system-conf.dtsi"
/{
xlnk {
compatible = "xlnx,xlnk-1.0";
};
};
```
What's more, I added 
```sh# CONFIG_BLK_DEV_INITRD is not set
CONFIG_XILINX_APF=y
CONFIG_XILINX_DMA_APF=y

CONFIG_CMA_SIZE_MBYTES=1024
CONFIG_STAGING=y

# CONFIG_CPU_IDLE is not set
# CONFIG_CPU_FREQ is not set
CONFIG_EDAC_CORTEX_ARM64=y
```
in project-spec/meta-user/recipes-kernel/linux/linux-xlnx/bsp.cfg.

After that I created SDSoC platform and tested matrix mult on it. However, I met "simple dma transfer failure accessing address 0xfe.....(I forgot the accurate address)" when executing the elf.
My ultra96 SDSoC platform was uploaded.

yunqu, 2018-09-20:
I am not sure what you did, are you creating your own image (since you are rebuilding your petalinux project)? Is your image able to boot without troubles?

For sdsoc flow on our image v2.3, the `pynqlib.c` file has to be compiled along with other source code:
https://github.com/Xilinx/PYNQ/blob/master/sdbuild/packages/libsds/libcma/pynqlib.c
https://github.com/Xilinx/PYNQ/blob/master/sdbuild/packages/libsds/libcma/libxlnk_cma.h

Again, sorry this is not documented yet. Then after you compile sdsoc project, get the `.so` file onto your board, along with bistream and tcl, and remember to set allocater like this:
https://github.com/yunqu/pynq-ekf/blob/master/ekf/ekf.py#L51

(However, I doubt this is the reason for your issue; but this is the right flow to make it work)

focalplane, 2018-09-26:
Hi jiangwx,  Are you building an overlay for the PYNQ framework to use?  If not you might want to ask this question in the Zedboard.org or the Xilinx forums.  Here is the Zedboard url:  http://zedboard.org/forums/ultra96-hardware-design

jiangwx, 2018-09-30:
> Hi jiangwx, Are you building an overlay for the PYNQ framework to use? If not you might want to ask this question in the Zedboard.org or the Xilinx forums. Here is the Zedboard url: http://zedboard.org/forums/ultra96-hardware-design

Thank You very much.

closed 2018-11-16 by focalplane