Avnet/Ultra96-PYNQ #13: Cannot open the jupyter page

aikenpang, 2018-12-22:
I download the 2.3 image to a 16GB SD card(which is the one come from the board) and flash it by Etcher.
The system boot up, all wifi, USB-ethernet and USB COM are working.
 I can ping 192.168.3.1 when I am using USB-ethernet but not in the browser.( Not able: Connecting to Jupyter Notebooks ). Same thing happen on WIFI connection with IP 192.168.2.1
 I can use USB COM to telnet into the board and I can use \\192.168.3.1\xilinx to access the files in the system.
BUT NOT …. Jupyter Notebooks.
Is there a way to check the Jupyter Notebook is running from the USB COM (serial connection)?

yunqu, 2019-01-02:
What about 
```shell
ps aux|grep jupyter
```
do you see a jupyter notebook process?

There should not be any problem with jupyter. I think it might be related to your network configuration of your browser or others. Maybe you dns settings is looking for that address in some other places rather than the local one. Try to disable other network connections on your PC to see if you can connect.

focalplane, 2019-01-26:
I hope this is working for you now!

closed 2019-01-26 by focalplane