Avnet/Ultra96-PYNQ #34: Failure to boot when using STM32 sensor mezzanine

skiphs, 2019-09-06:
I'm running into an issue when trying to boot my Ultra96-v2 with a PYNQ image while the STM32 sensor mezzanine is attached.

I've verified that PYNQ works with the "original" sensor mezzanine, as well as without one at all. However, if I try to attach the STM32 variant, the board doesn't boot. (No display, no USB serial port, the fan runs, but the 4 status LEDs all stay a solid green)

What's more it works perfectly fine when using the out-of-box image.

Mezzanine in question:
https://www.96boards.org/product/stm32/

focalplane, 2019-09-06:
I do not have an STM mezzanine to investigate this.

Are you using a 2A or 4A power supply?  If you can, use 4A. 2A is marginal even with no accessories attached.

Secondly, the #2 uart for pynq is different, it is coming from the PS and is outputing console to it.  There may be a conflict (I'm not familiar with the stm32) with it.  It can be changed in the Vivado hardware design but then PYNQ has to be rebuilt.

Those are my best guesses.

skiphs, 2019-09-06:
I'm using the 4A power supply, but your second suggestion was spot on!

I'm not sure why yet, but the board won't boot when UART1_RxD is connected to the STM mezzanine.
Thankfully, the board comes with a large amount of jumpers and dip switches. Breaking the solder bridge for JP60 disconnects UART1_RxD from the mezzanine and everything works.

I'll be sure to report back if I find a more suitable long-term solution, but this certainly solves my immediate problem. It's a great mezzanine if you can get past the fact it doesn't have a power button...

closed 2019-09-06 by skiphs

focalplane, 2019-09-07:
Very good that you have a work around in the short term.

This happened with the 1st pass of the Avnet Click mezzanine, the level translators resulted in the wrong electrical state on the RXD (RXD as input signal to the uarts).  A value of perpetual 'space' is an impossible situation resulting in an error 'break'.  I don't know if it is the uart itself or the software drivers talking to the uart that hang up under those conditions.  The only fix to this was to electrically drive the RXD input to a mark value when no data is flowing.  With the Click board, simply changing the PL pull-up strength could not fix it, a pull-up was added on Click board side of the level translators (it might have been down,  I don't recall and I don't have the schematic handy).

IF this is the case there is no way to make the STM32 work with PYNQ as-is.  My guess is that the out of box doesn't have a uart on those pins and that is why it doesn't hang, I'm pretty certain the out of box is routing the PS UART_1 to the side connector (that can attach the Avnet JTAG/serial adapter).  And nothing else is connected to the PL pins of the U96 to the low speed connector.

The benefit of Ultra96 and an FPGA (PL) is that it can do just about anything, pin functionality can be moved around quite a bit.  The downside is everybody tends to do a little of there own thing which can cause more than a little confusion sometimes.