Avnet/Ultra96-PYNQ #41: Ultra96 v2 & v2.5 - Boot OK but Done LED does not lit. up

nhphuong91, 2019-10-18:
I download the img from here: [http://www.pynq.io/board.html](http://www.pynq.io/board.html)
Everything is fine, except the indicator LED stuck at init LED (red one) and doesn't change to Done LED (blue one) even though the OS has loaded. This didn't happen when I use v2.4.
Hope that this issue only appear with the pre-built img on that webpage.

yunqu, 2019-10-18:
That is because in image v2.5, we do not load bitstream anymore during boot time. You will have to manually load bitstream to see DONE signal. The image will still work.

FredKellerman, 2019-10-18:
Hi nphuong91,

It is not an issue it is really just a missunderstanding.  That LED is not really a done booting LED it is really an indicator that a bitstream is loaded successfully into the PL of the SOC.  As yunqu has stated with v2.5 it does not load the PL with anything.  With PYNQ you can load your own bitstream / Overlay at any time and not have to reboot the system, as yunqu already said once you do it will go un-red.

I suppose a question might be: what if I want it to boot up with a bitstream how do I do that?  I don't know the answer to that or if that is your concern but I will ask it in the PYNQ forum ( https://discuss.pynq.io ).

Kind regards,
-Fred

focalplane, 2019-10-18:
The concern is normal behavior, closing issue.

closed 2019-10-18 by focalplane