Avnet/Ultra96-PYNQ #46: libxlnk_cma.h: No such file or directory

powderluv, 2019-11-12:
I am trying to build image v2.5 for Ultra96v2. After some wrestling with dependencies on my Ubuntu 18.04 system I am running into this error. Please suggest any workarounds. 

Ubuntu 18.04, Vivado, Petalinux and XSDK all are 2019.1.  (I tried 2019.2 but they have different issues and XSDK is removed)

End goal is to be able to build a custom kernel module that can drive the instantiation of the PL. 

    _xhdmi/hdmi_functions.c:5:10: fatal error: libxlnk_cma.h: No such file or directory                                                                         
     #include <libxlnk_cma.h>                                                   
              ^~~~~~~~~~~~~~~                                                   
    compilation terminated.                                                     
    embeddedsw_lib.mk:31: recipe for target '_xhdmi/libxhdmi.so' failed         
    make[2]: *** [_xhdmi/libxhdmi.so] Error 1                                   
    make[2]: Leaving directory '/tmp/pip-uzk59_qf-build/pynq/lib/_pynq'         
    Makefile:7: recipe for target 'all' failed                                  
    make[1]: *** [all] Error 2                                                  
    make[1]: Leaving directory '/tmp/pip-uzk59_qf-build/pynq/lib/_pynq/_xhdmi'  
    error: command 'make' failed with exit status 2                             
                                                                                
    ----------------------------------------                                    
  Can't roll back pynq; was not uninstalled                                     
Command "/usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-
uzk59_qf-build/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read
().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" instal --record /tmp/pip-3gw6phh4-record/install-record.txt --single-version-external
ly-managed --compile" failed with error code 1 in /tmp/pip-uzk59_qf-build/

FredKellerman, 2019-11-12:
Hi,  Ubuntu 16 LTS is used to build PYNQ not 18.04 LTS.  Did you follow the instructions in the README.md?  Where does it fail with respect to the order of those instructions?

For an example of building a custom kernel module look at how the wifi driver is built for U96 v2 (done using qemu).

powderluv, 2019-11-12:
It fails when I run:
make BOARDDIR=/home/USER/path/Ultra96-PYNQ

I can try with 16.04 but it seems unlikely it is a ubuntu version issue. 

Thanks for the tip on the wifi module - will try it out. I assume it uses qemu-user-static and rebuilds for the same kernel version and then we just insmod it.

powderluv, 2019-11-14:
I tracked down the problem further. Looks like the python packages are not being able to be installed because of a name resolution failure inside the chroot. I am trying a workaround to add "nameserver 8.8.8.8" in /etc/resolv.conf.  

libsds is installed after python so that fails to install and then you get the above error in your second try.

FredKellerman, 2019-11-14:
Good to hear you are moving forward!  There is no known issue if built with Ubuntu 16 LTS on any of my setups.

FredKellerman, 2019-11-14:
Another thought that just came to mind is that once a while back I tried to build off of my home dir which was LUKs encrypted.  The path nesting involved with PetaLinux exceed that filesystem's capacity and it failed in a weird way.  I don't recall the details.  Ever since I build on plain ext4 partitions like /sandbox/PYNQ and /sandbox/Ultra96-PYNQ.  I have also tried to push the envelop with newer versions of Ubuntu and the supporting libs/tools for the newer rootfs have had various issues.  I then just go back to good old 16.04/.10 LTS.  Sometimes the package server will not respond and a build fails but otherwise, again no issues.

focalplane, 2019-11-20:
No more correspondence.  Would be nice if 18 LTS could be used to build PYNQ as well as 16 LTS but right no one seems to have the time right now to take it the whole way.

closed 2019-11-20 by focalplane

powderluv, 2019-11-21:
Im traveling right now but in the next couple weeks I will upstream some changes to make it easy for 18.04+ releases

powderluv, 2019-12-05:
As promised in exactly 2 weeks :)

https://github.com/Xilinx/PYNQ/pull/974 adds support for 18.04 builds without the need for vagrant etc.

FredKellerman, 2019-12-06:
Thanks for the update!