Avnet/Ultra96-PYNQ #47: Build error when building PYNQ-compatible BSPs from scratch

masahi, 2019-11-15:
Hi, I'm trying to build PYNQ image along with BSP.  

I followed the "Building PYNQ-compatible BSPs from scratch" section in the README, but when I do "make BOARDDIR=\<LOCAL ULTRA96>", I got various errors like "undefined reference to `eth_register'". 

The error seems to be the same one reported [here](https://forums.xilinx.com/t5/Embedded-Linux/petalinux2019-2-u-boot-compile-error-for-ultra96-board/m-p/1039492). I tried the solution suggested in that post, but without success. 

Can you tell me how to resolve this issue?

My build environment: Ubuntu 16.04 on docker with Vivado and Petalinux 2019.1. I'm targeting Ultra96 v2.

focalplane, 2019-11-15:
Does the build work with the bsps in the repo?  I'd start there before making your own.

The error you reference is 2019.2 and using a different board configuration.  If using your own bsp make sure you configure it with the settings shown in the README.md.  V1 U96 works like a champ, V2 is more complicated but the errors you would get with troubles with it are not what you stated.

Also, you should make sure you can build your own bsp with just PetaLinux before you attempt to integrate it with U96 PYNQ.  If PetaLinux by itself can't build it then a PYNQ build has no chance.

masahi, 2019-11-16:
Thanks for the response. Yes, I tried building the image with the included bsp and it worked. I also tried "petalinux-build" inside Ultra96/bsp/sensors96b but got the same error.

I am trying to build my own bsp because I need to modify the cache coherency setting. I'm new to FPGA in general so I could be wrong, but from what I heard the PYNQ image of Ultra96 ships with hardware cache coherency disabled by default, but I need to enable it. 

So I followed [a blog post](https://qiita.com/okuhata/items/2557c50db77e14233ab0) (sorry it's in Japanese) and went ahead modifying PYNQ/sdbuild/Makefile and building everything including bsp from scratch.

I have two questions:
1. Do I indeed need to build my own bsp?
2. What makes things in V2 more complicated?

masahi, 2019-11-16:
I should mention that I'm using the v2.5 branch.

yunqu, 2019-11-16:
The error is because you have to include the u-boot patches:
https://github.com/Avnet/Ultra96-PYNQ/tree/master/Ultra96/petalinux_bsp_v1/meta-user/recipes-bsp/u-boot

For some reason the u-boot config in petalinux_bsp for v2 is missing. `sensors96b.bsp` has that u-boot included manually so that is working fine. But for your own bsp, either include the u-boot folder manually, or copy the u-boot folder under `petalinux_bsp_v2/meta-user/recipes-bsp/u-boot`. Then your make process should work.

masahi, 2019-11-16:
@yunqu Thanks, will try next week.

focalplane, 2019-11-16:
Good to hear that is solved. The issue of patches and trouble in the first place is really a Xilinx PetaLinux issue, they broke u-boot 2019.1 for Ultra96.  The confusion increases because we didn't pass along that you need the patches to build any Ultra96 v1 or v2 bsp for 2019.1.  Then it got a little worse because v1 and v2 were done separately and a little differently and the u-boot patch was not put in the PYNQ sidecar build folder.  That they require each a different bsp is another long conversation.  I will update the README.md a bit to indicate that you need the u-boot patch to work with the Ultra96 bsp, whether using PYNQ or not.

I just looked it all over and noticed also v2 for u-boot is missing a few networking features due to not having bsp.cfg as well included.  This should NOT really be an issue for anyone but v1 and v2 will be a little different there.

The reference to v2 bsp being more complicated: wifi for v2 brings in a kernel patch and recipe.  However those I did put into both the PYNQ add-on folders and the v2 bsp itself.

I do not know the answer to your root question: do you need to customize the bsp at all to change the cache coherency!  @yunqu ?

focalplane, 2019-11-16:
A PR was merged to clarify the u-boot issue and also place the patches in both places.  The redundant files for v2 were added on the **master** branch only but I did add some quick comments on the **image_v2.5** branch.  I'm closing the issue, I'm sure @yunqu will try to answer IF he knows about the cache coherency (whether it is disabled or not in PYNQ out of the box).

closed 2019-11-16 by focalplane

focalplane, 2019-11-17:
Just added the same files/readme to the **image_v2.5** branch also.