Avnet/Ultra96-PYNQ #51: mmc0 error during boot

ric96, 2019-11-18:
I am using the sensors96b_v2.bsp with petalinux and non of the pynq components.

Upon booting I can see the following dump
```
[  169.596187] mmc0: sdhci: ============================================
[  169.602674] mmcblk0: error -110 requesting status
[  179.728989] mmc0: Timeout waiting for hardware cmd interrupt.
[  179.734723] mmc0: sdhci: ============ SDHCI REGISTER DUMP ===========
[  179.741147] mmc0: sdhci: Sys addr:  0x00000000 | Version:  0x00001002
[  179.747571] mmc0: sdhci: Blk size:  0x00000000 | Blk cnt:  0x00000000
[  179.753994] mmc0: sdhci: Argument:  0x00000000 | Trn mode: 0x00000000
[  179.760417] mmc0: sdhci: Present:   0x01f70000 | Host ctl: 0x00000001
[  179.766840] mmc0: sdhci: Power:     0x0000000f | Blk gap:  0x00000080
[  179.773263] mmc0: sdhci: Wake-up:   0x00000000 | Clock:    0x0000fa07
[  179.779686] mmc0: sdhci: Timeout:   0x00000000 | Int stat: 0x00000001
[  179.786110] mmc0: sdhci: Int enab:  0x00ff0083 | Sig enab: 0x00ff0083
[  179.792533] mmc0: sdhci: AC12 err:  0x00000000 | Slot int: 0x00000001
[  179.798956] mmc0: sdhci: Caps:      0x31e8c881 | Caps_1:   0x00002007
[  179.805379] mmc0: sdhci: Cmd:       0x00000000 | Max curr: 0x00000000
[  179.811803] mmc0: sdhci: Resp[0]:   0x00000000 | Resp[1]:  0x00000000
[  179.818226] mmc0: sdhci: Resp[2]:   0x00000000 | Resp[3]:  0x00000000
[  179.824648] mmc0: sdhci: Host ctl2: 0x00000000
[  179.829076] mmc0: sdhci: ADMA Err:  0x00000000 | ADMA Ptr: 0x0000000000000000
[  179.836192] mmc0: sdhci: ============================================
```

and this is what xsct console shows:
```
xsct% XSDB Server Channel: tcfchan#9
Info: Cortex-A53 #3 (target 13) Running
xsct% Info: Cortex-A53 #3 (target 13) Running
xsct% Info: Cortex-A53 #0 (target 10) Stopped at 0x0 (Cannot resume. APB AP transaction error, DAP status 0x30000021)
xsct% Info: Cortex-A53 #3 (target 13) Running
xsct% Info: Cortex-A53 #3 (target 13) Running
xsct% Info: Cortex-A53 #2 (target 12) Stopped at 0x0 (Cannot resume. APB AP transaction error, DAP status 0x30000021)
Info: Cortex-A53 #2 (target 12) Running
xsct% Info: Cortex-A53 #3 (target 13) Stopped at 0x0 (Cannot resume. APB AP transaction error, DAP status 0x30000021)
Info: Cortex-A53 #3 (target 13) Running
xsct% Info: Cortex-A53 #1 (target 11) Stopped at 0x0 (Cannot resume. APB AP transaction error, DAP status 0x30000021)
Info: Cortex-A53 #1 (target 11) Running
xsct% Info: Cortex-A53 #3 (target 13) Stopped at 0x0 (Cannot resume. APB AP transaction error, DAP status 0x30000021)
xsct% Info: Cortex-A53 #3 (target 13) Running
xsct% Info: Cortex-A53 #3 (target 13) Stopped at 0x0 (Cannot resume. APB AP transaction error, DAP status 0x30000021)
Info: Cortex-A53 #3 (target 13) Running
xsct% Info: Cortex-A53 #3 (target 13) Stopped at 0x0 (Cannot resume. APB AP transaction error, DAP status 0x30000021)
Info: Cortex-A53 #3 (target 13) Running
xsct% Info: Cortex-A53 #3 (target 13) Stopped at 0x0 (Cannot resume. APB AP transaction error, DAP status 0x30000021)
Info: Cortex-A53 #3 (target 13) Running
xsct% Info: Cortex-A53 #1 (target 11) Stopped at 0x0 (Cannot resume. APB AP transaction error, DAP status 0x30000021)
Info: Cortex-A53 #1 (target 11) Running
xsct% Info: Cortex-A53 #3 (target 13) Stopped at 0x0 (Cannot resume. APB AP transaction error, DAP status 0x30000021)
xsct% Info: Cortex-A53 #3 (target 13) Running
xsct% Info: Cortex-A53 #3 (target 13) Stopped at 0x0 (Cannot resume. APB AP transaction error, DAP status 0x30000021)
Info: Cortex-A53 #3 (target 13) Running
xsct% Info: Cortex-A53 #1 (target 11) Stopped at 0x0 (Cannot resume. APB AP transaction error, DAP status 0x30000021)
xsct% Info: Cortex-A53 #1 (target 11) Running
xsct% Info: Cortex-A53 #3 (target 13) Stopped at 0x0 (Cannot resume. APB AP transaction error, DAP status 0x30000021)
Info: Cortex-A53 #3 (target 13) Running
xsct% Info: Cortex-A53 #1 (target 11) Stopped at 0x0 (Cannot resume. APB AP transaction error, DAP status 0x30000021)
xsct% Info: Cortex-A53 #1 (target 11) Running
xsct% Info: Cortex-A53 #1 (target 11) Stopped at 0x0 (Cannot resume. APB AP transaction error, DAP status 0x30000021)
Info: Cortex-A53 #1 (target 11) Running
xsct% Info: Cortex-A53 #1 (target 11) Stopped at 0x0 (Cannot resume. APB AP transaction error, DAP status 0x30000021)
Info: Cortex-A53 #1 (target 11) Running
xsct% Info: Cortex-A53 #1 (target 11) Stopped at 0x0 (Cannot resume. APB AP transaction error, DAP status 0x30000021)
Info: Cortex-A53 #1 (target 11) Running
xsct% Info: Cortex-A53 #1 (target 11) Stopped at 0x0 (Cannot resume. APB AP transaction error, DAP status 0x30000021)
xsct% Info: Cortex-A53 #1 (target 11) Running
xsct% Info: Cortex-A53 #1 (target 11) Stopped at 0x0 (Cannot resume. APB AP transaction error, DAP status 0x30000021)
Info: Cortex-A53 #1 (target 11) Running
xsct% Info: Cortex-A53 #1 (target 11) Stopped at 0x0 (Cannot resume. APB AP transaction error, DAP status 0x30000021)
Info: Cortex-A53 #1 (target 11) Running
xsct% Info: Cortex-A53 #1 (target 11) Stopped at 0x0 (Cannot resume. APB AP transaction error, DAP status 0x30000021)
xsct% Info: Cortex-A53 #1 (target 11) Running
xsct% Info: Cortex-A53 #1 (target 11) Running
xsct% Info: Cortex-A53 #1 (target 11) Stopped at 0x0 (Cannot resume. APB AP transaction error, DAP status 0x30000021)
xsct% Info: Cortex-A53 #1 (target 11) Running
xsct% Info: Cortex-A53 #1 (target 11) Stopped at 0x0 (Cannot resume. APB AP transaction error, DAP status 0x30000021)
Info: Cortex-A53 #1 (target 11) Running
xsct% Info: Cortex-A53 #1 (target 11) Stopped at 0x0 (Cannot resume. APB AP transaction error, DAP status 0x30000021)
Info: Cortex-A53 #1 (target 11) Running
xsct% Info: Cortex-A53 #1 (target 11) Stopped at 0x0 (Cannot resume. APB AP transaction error, DAP status 0x30000021)
Info: Cortex-A53 #1 (target 11) Running
xsct% Info: Cortex-A53 #1 (target 11) Stopped at 0x0 (Cannot resume. APB AP transaction error, DAP status 0x30000021)
Info: Cortex-A53 #1 (target 11) Running
xsct% Info: Cortex-A53 #1 (target 11) Stopped at 0x0 (Cannot resume. APB AP transaction error, DAP status 0x30000021)
xsct% Info: Cortex-A53 #1 (target 11) Running
xsct% Info: Cortex-A53 #1 (target 11) Stopped at 0x0 (Cannot resume. APB AP transaction error, DAP status 0x30000021)
Info: Cortex-A53 #1 (target 11) Running
xsct% Info: Cortex-A53 #1 (target 11) Stopped at 0x0 (Cannot resume. APB AP transaction error, DAP status 0x30000021)
Info: Cortex-A53 #1 (target 11) Running
xsct% Info: Cortex-A53 #1 (target 11) Stopped at 0x0 (Cannot resume. APB AP transaction error, DAP status 0x30000021)
Info: Cortex-A53 #1 (target 11) Running
xsct% Info: Cortex-A53 #1 (target 11) Stopped at 0x0 (Cannot resume. APB AP transaction error, DAP status 0x30000021)
Info: Cortex-A53 #1 (target 11) Running
xsct% Info: Cortex-A53 #1 (target 11) Stopped at 0x0 (Cannot resume. APB AP transaction error, DAP status 0x30000021)
xsct% Info: Cortex-A53 #1 (target 11) Running
xsct% Info: Cortex-A53 #1 (target 11) Stopped at 0x0 (Cannot resume. APB AP transaction error, DAP status 0x30000021)
xsct% Info: Cortex-A53 #1 (target 11) Running
xsct% Info: Cortex-A53 #0 (target 10) Running
```

I am not able to boot into linux.

ric96, 2019-11-18:
Also it looks like it only happens when the jtag pod is connected and vitis or sdx interacts with it say running GDB

focalplane, 2019-11-18:
Hi, Try a different SD card, Delkin or Sandisk.  Samsung EVO cards I know do not work.  I have used the jtag pod with the serial port with that bsp and pynq and saw no issues.  Using GDB as you said is likely the culprit for the 2nd case.  Sorry, no experience using that BSP with GDB and the jtag pod.  

Also, if you are using Vitis, that is using 2019.2 tools and PYNQ v2.5 is using 2019.1 which may or may not work properly.

focalplane, 2019-11-20:
No more information or reply, closing.

closed 2019-11-20 by focalplane