Avnet/Ultra96-PYNQ #52: Ultra96 V2 2.5 Bluetooth not working

AnantaBalaji, 2019-11-19:
Hi,

In the ultra96 V2 2.5 image, I could not use bluetooth. I need to use bluetooth for my project. 

Whenever I use hciconfig or hcitool , No devices are being found. Can you please let me know how can I enable bluetooth in the ultra96 v2 board?

focalplane, 2019-11-19:
Hi, We will look into this.  Possibly the easiest thing in the short term might be to use a USB bluetooth adapter.

AnantaBalaji, 2019-11-19:
Hi. USB Bluetooth adapter does work. But I would like to use Internal bluetooth already present in ultra96 v2 board. Please do let me know soon. I have been trying to bring it up for the past week.

focalplane, 2019-11-19:
Hi, This gets the driver up and running.  I don't really know much about all the modes and settings for bluetooth and what exactly this part supports but the shell commands below do get the driver up and running!

If you get the rest of it working please do share, I'm sure other people would like to know more as well.  So would I but right now I don't have time to dig in further.  Good luck! 

```shell
# Do this only once:
echo BT_POWER_UP > /dev/wilc_bt
echo BT_DOWNLOAD_FW > /dev/wilc_bt
echo BT_FW_CHIP_WAKEUP > /dev/wilc_bt

stty -F /dev/ttyPS1 115200
stty -F /dev/ttyPS1 crtscts
# Initialize the device:
hciattach /dev/ttyPS1 -t 10 any 115200 noflow nosleep
sleep 1s
hciconfig hci0 up
hciconfig hci0
```
Final command output:
```
hci0:   Type: Primary  Bus: UART
        BD Address: F3:e0:33:C4:22:2F  ACL MTU: 27:30  SCO MTU: 0:0
        UP RUNNING 
        RX bytes:488 acl:0 sco:0 events:38 errors:0
        TX bytes:536 acl:0 sco:0 commands:38 errors:0
```
Related dmesg output:
```
[   78.375059] at_pwr_dev: open()
[   78.378216] AT PWR: bt_power_up
[   78.378232] wilc_sdio mmc1:0001:1: SDIO speed: 50000000
[   78.378510] wilc_sdio mmc1:0001:1: chipid 003000d0
[   78.378514] WILC POWER UP
[   78.381171] at_pwr_dev: close()
[   78.382436] at_pwr_dev: open()
[   78.385548] AT PWR: bt_download_fw
[   78.385552] Bluetooth firmware: mchp/wilc3000_ble_firmware.bin
[   78.402123] Downloading BT firmware size = 58184 ...
[   78.557697] Starting BT firmware
[   78.663974] BT Start Succeeded
[   78.664212] at_pwr_dev: close()
[   78.665639] at_pwr_dev: open()
[   78.669979] at_pwr_dev: close()
```

AnantaBalaji, 2019-11-20:
Thanks a lot. It does work now. I did not mention the initialization timeout in my hciatatch commands.

closed 2019-11-20 by focalplane