Avnet/Ultra96-PYNQ #55: none of the pre-built images are

mdxtinkernick, 2020-03-22:
trying to download any of the prebuilt images produces a page with this text

Third-party server not responding.
The resource you have requested is located on a third-party server. WebSEAL has attempted to send your request to that server, but it is not responding. 
Explanation
This could be due to the third-party server being offline, or to network problems making it unreachable. The problem is not with the WebSEAL server itself.
Solutions
Retry your request later, or contact the system administrator for assistance.
[none BUTTON]

not sure if you are the system admin for where it is stored, but flagging it here incase

yunqu, 2020-03-22:
I think you are right. Thanks for letting us know; we will look into it.

zedhed, 2020-03-24:
@mdxtinkernick, the Avnet docs site is back up and these files should be available for you to download now.  Can you please confirm on your end?

Thank you for bringing this to our attention, sorry for any inconvenience this may have caused you! 

-Kevin

closed 2020-03-25 by zedhed

mdxtinkernick, 2020-03-25:
yes working fine now, thanks