Avnet/Ultra96-PYNQ #57: AXI4 master in PL

combinatorylogic, 2020-05-01:
Is it possible to enable, say, S_AXI_HPC0_FPD without rebuilding the BSP? If not, can we have it enabled by default / at least have an official build where it's enabled and driven by some stub?

yunqu, 2020-05-01:
You should be able to do that by checking ZYNQ Ultrascale registers. In `pynq` there is also a `Register` class that helps you to change register settings.

combinatorylogic, 2020-05-01:
Great, thanks! Will it work if I simply load an overlay with it enabled?

yunqu, 2020-05-01:
It should. We have done that for a couple of cases where we need to change the AXI port data width on PS side. Also if you may have noticed we used `Register` to change the 4 PL clocks.

combinatorylogic, 2020-05-01:
Ok, great, thanks! That's nice. I extrapolated the experience with Cyclone-V and its memory ports configuration, glad Xilinx allows a runtime reconfiguration.

closed 2020-05-01 by combinatorylogic