Avnet/Ultra96-PYNQ #60: How do I get connected to the board (Ultra96-v2 & Ultra96-PYNQ v2.5)?

fengyuentau, 2020-08-15:
Hi, I got the Ultra96-v2 board from the contest DAC-SDC 2020, and it was shipped to me with the board itself, an SD card and an adapter, and several AC adapters. It was because of the COVID-19 that I can only get my hands on the board recently. I am new to embedded devices and FPGA so I need to learn a lot of things.

Firstly, I followed the instructions on the getting-started guide from the package box and everything works great with the `ultra96v2_oob_2018_3_190917_8GB.img` image. I tried several examples and toggled the WIFI settings.

But when I flashed another image `ultra96v2_v2.5.img` from http://www.pynq.io/board.html, problems occurred: I cannot get connected with the board through neither USB port, WIFI or the miniDP. Turn on the board and LEDs are:
- Done LED (D1): off
- Power On LED (D2): on
- 4x User LEDs: D3/4/6: off, D7: yellow light in heartbeat pattern
- Wireless LAN (D9) and Bluetooth (D10) LEDs: on

In the very beginning, I thought the problem was the red LED, but the fact is not true since https://github.com/Avnet/Ultra96-PYNQ/issues/41#issuecomment-543810045 explains that the red LED is because of the non-loaded bitstream. I don't know what the bitstream stands for though.

I tried the following connecting methods, but all failed:
- Connecting with the USB port. I learned from https://www.element14.com/community/docs/DOC-93372/l/ultra96-v2-gsg-v11?ICID=ultra96v2-datasheet-widget that it seems the v2 board can only be able to get connected via USB connection using the USB-to-JTAG/UART pod, but then I went to the contest discussion platform https://piazza.com/class/k3om932nrqxcj and found that some contesters were able to connect via USB without the Pod. Do I need to install Vivado or something in order to connect with the board via USB?  My OS is windows 10 BTW.
- Connecting with WIFI. I learned from https://github.com/Avnet/Ultra96-PYNQ/issues/33#issuecomment-525821080 and https://pynq.readthedocs.io/en/latest/faqs.html?highlight=wifi%20access%20point#how-can-i-enable-wireless-access-point that the WIFI access is a feature for PYNQ that needs to be turned on manually. How do I turn on this feature in the case I cannot get connected with the board? 
- Setting up a local desktop with a monitor, mouse and keyboard. There is just simply no output signal when I connect the monitor to the board's miniDP port using a miniDP-to-HDMI cable, even if the click mouse and type on the keyboard.

So what should I do to get connected with the board using PYNQ? I am installing Vivado and HLS. Hope this will help me get connected via USB.

fengyuentau, 2020-08-15:
By the way, nothing shows up when I connect the board with my windows 10 desktop and my macbook. No unknown devices were shown up in the Device Manager. I can confirm that both the USB ports on my desktop and the miniDP-to-HDMI cable works normally.

fengyuentau, 2020-08-15:
Problem solved. I was misled by the image showing the micro-usb cable:

[image]

I managed to get connected to the board with the following port (green-tick).
[image]

closed 2020-08-15 by fengyuentau

FredKellerman, 2020-08-17:
You are likely using a micro-usb 2.0 cable that only supports power and not the data signals or the cable is defective or not compatible.  Some cables are designed to only charge with and do not have all the pins, if I look closely at your photo I can only see 2 pins..  The diagram as shown in the instructions is correct as long as you have working full micro usb cable.  You do not have to use a micro USB 3.0 cable either.

happysheep224, 2020-11-15:
Have you solve your problem? My win10 computer cannot search the  pynq_<MAC Address>

royliuyu, 2021-04-14:
Your success inspired me, I met exactly the same issue: Can not use OTG cable connect the USB port. Now I got the answer. 
My goodness, this video also mislead me...