Avnet/Ultra96-PYNQ #64: Route UART1 to expansion connector

jan-kiszka, 2020-09-15:
What is missing in the default BSP to route UART1 also to the pins on the expansion connector? That way, the UART-USB converters on shields like https://www.96boards.org/product/uartserial/ or https://www.96boards.org/product/sensors-mezzanine/ would just work for default console, like on other 96boards.

Back then, Alex Graf did this for the Ultra96-v1 BSP that OpenSUSE used to provide, and I adopted it for https://github.com/siemens/jailhouse-images. I'm missing this now in the v2 BSP which I had to build from this repo. Would be nice to keep it in my image, and I could also imagine that users of the reference image would appreciate such a feature.

focalplane, 2020-09-15:
Internally we've gone back and forth with UART mappings, it was a compromise with traditional 96boards, Xilinx devices, PetaLinux and usage of the USB to JTAG / UART add on board that Avnet makes.  The grove mezzanine does work with the Ultra96 but not as the default console, so yes other applications would need re-compiled or different output specified if supported.

You're welcome to do a pull request but... at this point changing UARTs again is not likely to happen unless there would be strong demand from users to do that.  You are the first to request it and we do thank you for your good intention.  If this UART you ask for would not move any other UARTs that would be easier to consider.

jan-kiszka, 2020-09-15:
I was wondering if the wouldn't be a way to route the UART to both connectors in parallel. Effectively, I got that working for the output by mixing Alex' bitstream with the official fsbl. But input does not work then, possibly because of wiring or fsbl not accounting for parallel usage. Or it's really a fundamental conflict that can't be resolved. I'm speculating here - no expertise in FPGA designing.

focalplane, 2020-09-16:
Ignoring the software side of managing the uart(s).  The Xilinx SoC cannot route all combinations of Uarts and external I/O pins together internally.   Some of the uarts do not connect to or through the fpga and those are the same as any other non fpga based SoC.  If you implemented uarts in the programmable logic there is a lot more flexibility in how things can be tied together.  The Ultra96 board SoC pins that are wired to the expansion headers are a mix of fpga and non-fpga peripherals and mostly the latter.

closed 2020-09-16 by focalplane