Avnet/Ultra96-PYNQ #79: Ultra96-v2 RNDIS Driver fails to install on Windows 10 64-bit

jomoengineer, 2021-04-04:
I am attempting to connect a Ultra96-V2 running v2.6 PYNQ for Ultra96, however when connecting the Ultra96-V2 to a Windows 10 64-bit system via the USB cable, the RNDIS driver fails to load.

This is the error from Device Manager:
```
The drivers for this device are not installed. (Code 28)

There are no compatible drivers for this device.

To find a driver for this device, click Update Driver.
```

The USB Controllers entry shows Unknown for the USB Device from the Ultra96-V2 with this under Device Status:
```
Windows has stopped this device because it has reported problems. (Code 43)
```

This is the message from the Ultra96-V2 dmesg:
```
[   11.066978] usb0: HOST MAC 22:ca:55:3d:96:df
[   11.067049] usb0: MAC 62:85:54:cb:43:2d
[   11.371016] configfs-gadget gadget: high-speed config #1: c
[   11.371310] IPv6: ADDRCONF(NETDEV_CHANGE): usb0: link becomes ready
[   11.505596] random: crng init done
[   11.505610] random: 7 urandom warning(s) missed due to ratelimiting
[   13.825323] [drm] Pid 510 opened device
[   13.825345] [drm] Pid 510 closed device
[   14.291373] [drm] Pid 510 opened device
[   14.291507] [drm] Pid 510 closed device
[ 1809.886629] configfs-gadget gadget: high-speed config #1: c
```

The PC connection is USB-2 so could that be the issue?

I can connect a BeagleBone Black to the same system without issue.

focalplane, 2021-04-09:
RNDIS works well on Windows and Linux when connected to Ultra96 PYNQ.  Not sure what is going on with your setup.  USB 2.0 vs 3.0 cable is not a problem.  Assuming your U96 is working correctly, try different cables.  There could also be some other driver installed on your PC causing some kind of conflict.  Try a different PC.

jomoengineer, 2021-04-09:
It works fine with a LapTop that has USB 3.0 and with an ODROID-C4 using the same cable.  The same PC that has an issue with the Ultra96-V2 and PYNQ works fine running RNDIS with a BeagleBone so I am not sure why there is an issue with the Ultra96-V2.

focalplane, 2021-04-09:
Don't know, I usually use USB 3.0 host ports and have no issues.  If it is a USB 2.0 host port compatibility issue, not saying it isn't possible, but you are the first person ever to question it publicly.  PYNQ for U96 has been around since 2018.

I'd be more inclined to think it is that some other RNDIS driver version was installed on your PC for your other boards and the driver activates and then tries to unsuccessfully talk to the Ultra96.  That is just a guess, USB has many layers and ways to get into trouble.  I don't have a USB 2.0 port to conveniently try out right now.

jomoengineer, 2021-04-09:
On the same PC that I am using which has USB 2.0, with a VirtualBox Ubuntu 20.04 VM with USB set to 2.0, the RNDIS driver does load and the USB ethernet device is created with an assigned ip of 192.168.3.100.
```
[  434.535559] usb 1-1: new high-speed USB device number 2 using ehci-pci
[  434.810900] usb 1-1: New USB device found, idVendor=1d6b, idProduct=0104, bcdDevice= 1.00
[  434.810903] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  434.810905] usb 1-1: Product: PYNQ-USB
[  434.810907] usb 1-1: Manufacturer: Xilinx Inc.
[  434.810908] usb 1-1: SerialNumber: 0000
[  435.026140] cfg80211: Loading compiled-in X.509 certificates for regulatory database
[  435.035314] cdc_acm 1-1:1.2: ttyACM0: USB ACM device
[  435.040973] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[  435.044005] usbcore: registered new interface driver cdc_acm
[  435.044008] cdc_acm: USB Abstract Control Model driver for USB modems and ISDN adapters
[  435.049848] usbcore: registered new interface driver cdc_ether
[  435.105287] rndis_host 1-1:1.0 usb0: register 'rndis_host' at usb-0000:00:0b.0-1, RNDIS device, 12:50:fc:3e:75:c9
[  435.119137] usbcore: registered new interface driver rndis_host
[  435.136638] usbcore: registered new interface driver rndis_wlan
[  435.170136] usb-storage 1-1:1.4: USB Mass Storage device detected
[  435.183189] scsi host3: usb-storage 1-1:1.4
[  435.183523] usbcore: registered new interface driver usb-storage
[  435.190672] usbcore: registered new interface driver uas
[  436.202901] scsi 3:0:0:0: Direct-Access     Linux    File-Stor Gadget 0504 PQ: 0 ANSI: 2
[  436.203397] scsi 3:0:0:0: Attached scsi generic sg2 type 0

```

This is similar to what I had seen with the ODROID-C4 device which is also running Ubuntu 20.04.

However, back on the Windows 10 64-bit system, after connecting the Ultra96-V2 via USB, I do see the device gets mounted and I can navigate to the files on the Ultra96-V2.

Opening a command prompt in Windows, I changed into the PYNQ-USB device and then in the serial_driver directory, I see two 'linux-cdc-acm' files.
```
J:\>cd serial_driver

J:\serial_driver>dir
 Volume in drive J is PYNQ-USB
 Volume Serial Number is 7110-7329

 Directory of J:\serial_driver

10/20/2020  10:08 AM    <DIR>          .
10/20/2020  10:08 AM    <DIR>          ..
10/20/2020  10:08 AM             4,118 linux-cdc-acm.cat
10/20/2020  10:08 AM             3,492 linux-cdc-acm.inf
               2 File(s)          7,610 bytes
               2 Dir(s)      52,011,008 bytes free
```
If I try to run the .cat file, I get the following error that shows up.
```
J:\serial_driver>linux-cdc-acm.cat
```
[image]

I changed the cable and I get the same result on this Windows 10 system.

I believe this is where it is getting hung up at.

jomoengineer, 2021-04-10:
Yeah, I am thinking as you mentioned that there is some sort of conflict with an existing driver on my Windows 10 PC that is interfering with the RNDIS driver for the Ultra96-V2.
Even with a Petalinux build and running the 'usb_gadget_ethernet.sh' script that comes with the Petalinux build, I get the same issue with the RNDIS driver on Windows and the script just hangs at this point:
```
usb-gadget-ethernet.sh   usb-gadget-ethernet.sh~  usb_gadget_ethernet.sh
root@ultra96v2-2020-1:~# ./usb_gadget_ethernet.sh
[   70.263110] using random self ethernet address
[   70.267593] using random host ethernet address
[   70.309793] usb0: HOST MAC 0a:6b:f7:f2:bf:5c
[   70.314294] usb0: MAC 26:fa:fe:8f:6d:72
**************************************
Assigned static IP 192.168.3.1 to usb0
**************************************
root@ultra96v2-2020-1:~# [   70.635350] configfs-gadget gadget: high-speed config #1: c
[   70.641122] IPv6: ADDRCONF(NETDEV_CHANGE): usb0: link becomes ready

```

I'll have to see if I can track down what is causing this on the system.

Thanks.

closed 2021-05-04 by focalplane