Avnet/Ultra96-PYNQ #81: Failed to install docker _not sure if it is OS incompatible issue

royliuyu, 2021-04-21:
I am going to install docker and then use FINN.
But encounter some errors:
I follow the install instruction via [Install Docker Engine on Ubuntu](https://docs.docker.com/engine/install/ubuntu/), in order to fit the ubuntu contribution codename  with docker's, I revised the codename from WFH to bionic in file of "lsb_release".
But when I install by using command :
[image]

There is still errors like"

[image]

Is anyone has experienced and gets any solution?

schelleg, 2021-04-21:
Can you try steps here:
https://discuss.pynq.io/t/docker-xilinx-platforms-pynq/1962

we hardcode versions of those deb packages and have verified docker installs correctly.

royliuyu, 2021-04-21:
schelleg，
Your URL really works, I amount docker with the guidence successfully.
Thanks

closed 2021-04-21 by royliuyu