Avnet/Ultra96-PYNQ #85: unable to get grove sensors96 mezannine working with Ultra96

ProjectHydroTech, 2021-12-08:
link to board: https://www.96boards.org/product/sensors-mezzanine/ 

I managed to generate a sensors96 bitstream in vivado to install into the Ultra96 as the overlay, I am able to access the PL/PS ports but i am unable to communicate to the onboard arduino in any way. Output of flashing Blink.ino here:
[image]

Running dmesg | grep tty to check the available serial ports gives me this output but i am unable to communicate to them
[image]

In case my bitstream is wrong here are the steps:
i cloned the sensors96b directory, removed the v2 in sensors96b.tcl.v2, opened vivado tcl and ran source ./sensors96b.tcl to which a block diagram opened for me to generate HDL wrapper and then generated the bitstream where i copied the .bit and .hwh files to the Ultra96. This is the block diagram
[image]

focalplane, 2021-12-16:
Hi,  I am not sure what the issue is with your custom design.  I looked at your design very quickly, I do not see anything wrong on the surface.

Also, to use that .tcl file, be sure to invoke all of the steps that are documented in the makefile: https://github.com/Avnet/Ultra96-PYNQ/blob/master/Ultra96/sensors96b/makefile

Have you tried the built-in sensors96 notebook & bitstream and verified that it works?  Referring to this one: https://github.com/Avnet/Ultra96-PYNQ/blob/master/Ultra96/sensors96b/notebooks/sensors_mezzanine_examples.ipynb

Kind regards

focalplane, 2021-12-16:
Hi, One more thought: when PYNQ boots it does not load any bitstream into the FPGA/PL.  The PetaLinux driver does exist for the UART but you have to load the bitstream if you want it to work.

Kind regards

closed 2021-12-21 by focalplane

ProjectHydroTech, 2021-12-21:
Thanks for the reply. I have tried the steps in the makefile but the bitstream still does not work.
I have tried the notebook and everything in it does work but it doesnt provide example code for me to test the arduino portion of the sensors mezannine. I did try uploading the standard blink.ino for arduinos but was unable to do so due to the afformentioned problem.
I do see the PetaLinux driver is installed which i assume is for the uart 16550 and i did load the bitstream from the top of the notebook.