Avnet/Ultra96-PYNQ #86: V2.7 PYNQ for Ultra96v2, cannot import pynq

dj-park, 2021-12-08:
```
Python 3.8.2 (default, Mar 13 2020, 10:14:16) 
[GCC 9.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import pynq
/usr/local/share/pynq-venv/lib/python3.8/site-packages/pynq/pl_server/xrt_device.py:88: UserWarning: xbutil failed to run - unable to determine XRT version
  warnings.warn(
>>> from pynq import Xlnk
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: cannot import name 'Xlnk' from 'pynq' (/usr/local/share/pynq-venv/lib/python3.8/site-packages/pynq/__init__.py)
>>> 
```

Hi, I am using the recent v2.7 image for Ultra96 v2. I wrote the image to sd card and ssh to the board.
I opened up the python interactive session by typing python, and am trying to import pynq and xlnk, it doesn't work.
I was able to use v2.5 with Ultra96v2 normally.

focalplane, 2021-12-16:
Hi,

Xlnx for PYNQ v2.7 has been deprecate.  This is not unique to Ultra96.  If you must use that feature, please use an earlier version of PYNQ.

Please see here for more info:  https://github.com/Xilinx/PYNQ/releases/tag/v2.7.0

Kind regards

focalplane, 2021-12-16:
Hi,

About 'import pynq', it is a long story but in this release in order for PYNQ to be compatible with other platforms it was moved to a virtual environment.  You can see this in the warning message path 'pynq-venv'.

I assume when you ssh into the board you did so as the user xilinx (or some other user you added), try running the Python interpreter as sudo and see if that helps.

Kind regards

dj-park, 2021-12-19:
If I sudo python, is says
`ModuleNotFoundError: No module named 'pynq'`

If I don't do sudo, it complains when I do(import pynq passes..)
overlay = Overlay("./my_bitstream.bit")

`OSError: Root permissions required.`

focalplane, 2021-12-19:
Hi, You must have done something different than before if import pynq now works :-)

Did you make sure the .bit you are using and folder it is in has user privileges?

Kind regards

dj-park, 2021-12-21:
What I meant is,
`sudo python my_python.py`
results in `ModuleNotFoundError: No module named 'pynq'` like below

[image]

`python my_python.py`
results in `OSError: Root permissions required` like below.
[image]

I did `chmod 777 my_bitstream`

focalplane, 2021-12-21:
Hi, Thanks for clearing up what you meant.  As I mentioned prior, a virtual environment is now used for PYNQ.  Mainly so that it is compatible with some of the other Xilinx boards that directly run full Canonical Ubuntu vs. PYNQ that uses a custom ubuntu rootfs.  All platforms now use PYNQ with the venv v2.7.

See release notes here:  https://github.com/Xilinx/PYNQ/releases/tag/v2.7.0

I don't have a board up and running to verify and give you exact syntax but basically when you are logged into the terminal you need to activate the pynq virtual environment venv then try running your code.

This is not actually an Ultra96 issue, it is a new behavior of PYNQ v2.7 in general.  Unfortunately I am not aware of any documentation that explains this in more detail at this time.  Otherwise I would point you to it.

Official PYNQ support and discussion can be found under Support here:  http://www.pynq.io

I am sure many others will have similar issues and it would be good to post your questions and discuss it on the official forum so that it can benefit others too.

I hope that helps!

Kind regards

dj-park, 2021-12-21:
Thanks for the prompt responses.
So I managed to get it working.
Because of the permission issue, I logged in as root, and then activate pynq-venv.
```
sudo -s
source /usr/local/share/pynq-venv/bin/activate
python my_python.py
```

closed 2021-12-26 by dj-park