Avnet/Ultra96-PYNQ #97: yocto support

embetrix, 2024-07-30:
any plans to support yocto specially with the **ultra96-v2** board ?

the ultra96-zynqmp seems to be already supported :

https://github.com/Xilinx/meta-xilinx/blob/master/meta-xilinx-vendor/conf/machine/ultra96-zynqmp.conf

focalplane, 2024-07-31:
Hi,  This is the PYNQ repo, we use whatever tools and workflow AMD decides to use.  The latest release uses PetaLinux which is a very useful tool layer for AMD-Xilinx SoCs on top of the Yocto framework.

Why are you requesting Yocto?  Knowing why may influence future choices for non-PYNQ U96 bsps.

Closing this, it is not a PYNQ U96 issue.  But do feel free to carry on the conversation here.

embetrix, 2024-07-31:
Petalinux is not the first choice of developing a products, since it is a niche.

I think I can do it myself.

For that I need just a Hardware Description file, can you please point me where can I find a Design file *.XSA for the ultra96-v2  ?

The DTSI is already included here in this repository.

focalplane, 2024-07-31:
The XSA is created by using the build script in the repository.  There are no plans to deviate from what AMD does with PYNQ.

closed 2024-07-31 by focalplane